ros-industrial/abb #79: MoveIt! path planning superfluous

Gengoz, 2015-04-09:
Currently I am using MoveIt! to plan motion paths for the irb120. It works fine and uploads the 20 or so points to the robot, which then performs the movements. 

The irb120 can move fine on it's own though, without the help of ROS or MoveIt!. So I feel as if MoveIt! plans the path, sends the points to the ABB controller, which then uses the points to replan it's own path. If this is so, wouldn't it be possible to just send a single point to the robot and let the robot controller (IRC5) figure out it's own path? 

Is the path provided by MoveIt! in most cases superfluous and only usefull in situations where an object not known to the robot itself needs to be avoided?

I hope my question is clear.

With kind regards,

Tim

shaun-edwards, 2015-04-09:
@Gengoz, MoveIt! and ROS are useful for avoiding collisions with objects in the environment and with the robot itself (including the tool).  If you are just performing simple robot motion, with no need for collision avoidance, then ROS may not be the right tool.  If you still want to control the robot externally, you can use the ROS interface and simply send a trajectory with the current joint position and the desired joint position.  You can use the IK service provided by MoveIt! if you need to translate a Cartesian pose into a joint pose.

gavanderhoorn, 2015-04-15:
> Is the path provided by MoveIt! in most cases superfluous [..]

I'd say *most cases* entirely depends on which cases you are considering :). From a traditional automation perspective you could say that, but that would be ignoring the actual usefulness of sensor-based motion planning, which is something that I'd say is at the core of ROS and one of the reasons that ROS-Industrial exists.

Depending your perspective, you may think of the IRC5 performing 'just' the *interpolation* between two consecutive trajectory points. Obviously at some level it is actually *planning* (as it needs to determine how to get from where it is to where you want it to go), but I personally reserve the term *planning* for something that has more situational awareness (which is I think is what @shaun-edwards was referring to), such as MoveIt.

Example: the IRC5 will happily plan a path between two points on two opposite sides of a wall, without taking into account it cannot physically execute that motion. Only what is typically called a 'higher level' planner which has access to a world model would figure out it needs to *insert additional points* to get the manipulator safely from one side to the other.

If you just want to get your TCP from point A to point B and know a priori that is always possible (or don't care about possible collisions), then you obviously have no need for any contextual or sensor-based planning, and can just send those points to the controller directly.

Gengoz, 2015-04-16:
That cleared things up. Thank you both very much!

closed 2015-04-16 by shaun-edwards