ros-industrial/abb #80: irb2400 IKFast plugin doesn't match URDF

JeremyZoss, 2015-05-21:
URDF `tool0` definition was updated in 24f5530df3aec9e99bfcd764178b612ee7e7c5b7, but ikFast analytic solver was not updated.  IKFast solver now produces results inconsistent with URDF/KDL.

**Test case**
Compute IK for robot at nominal (all-zeros) position:
```
rosservice call /compute_ik "{ik_request: {group_name: 'manipulator', robot_state: {joint_state: {name: ['joint_1', 'joint_2', 'joint_3', 'joint_4', 'joint_5', 'joint_6'], position: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]}}, ik_link_name: 'tool0', pose_stamped: {pose: {position: {x: 0.94, y: 0.0, z: 1.455}, orientation: {x: 0.0, y: 0.707106781185, z: 0.0, w: 0.707106781188}}}}}"
```
 * KDL: `[0, 0, 0, 0, 0, 0]`
 * IKFast: `[0, 0.15, -0.28, 0.0, 1.70, 0.0]`

Other ABB models are not affected, because no IKFast plugin has been created.

shaun-edwards, 2015-05-21:
@Levi-Armstrong, this needs to be addressed ASAP, as it is in a released package.

gavanderhoorn, 2016-04-08:
Wasn't this fixed with the merge of #81?

JeremyZoss, 2016-04-13:
I see that fixes were applied to devel + release packages:
  * fixes in #81 (indigo-devel), #83 (indigo), #84 (hydro-devel), #85 (hydro)

But when I try to run my test-code above (with the latest indigo packages), I get different answers from KDL and IKFast plugins.  This is just a quick test, so I could be doing something wrong.  Dragging the robot in RVIZ (e.g. Tool-Z) causes the robot to flip to a different elbow configuration near the all-zeros position, when using the IKFast plugin, but not when using the KDL plugin.

It seems to me that the above PRs fix this particular bug.  There may be another bug where the IKFast plugin does not return the closest solution??  Or, perhaps there's a rounding error that causes IKFast to miss the solutions to the particular cartesian position in my test case.

I'm probably okay to close this bug.  If someone identifies the KDL/IKFast configuration-discrepancy as a "real" issue later, then we can open a new bug for that topic.

@Levi-Armstrong : do you concur?

Levi-Armstrong, 2017-01-10:
I believe we have been using this on our 2400 in the lab but I will make sure it is this version and not a custom one before closing the Issue.

gavanderhoorn, 2017-01-26:
@Levi-Armstrong: have you had an opportunity to test this?

Levi-Armstrong, 2017-01-27:
Not yet, I should have an update tomorrow. @Jmeyer1292

Levi-Armstrong, 2017-02-06:
@Jmeyer1292, Have you gotten a chance to check this?

Jmeyer1292, 2017-02-06:
It's not the same. On Godel we're using a version that was generated with a "newer" IK-Fast.

commit 9c9b76e3 (2025-04-05): irb2400: delete IKFast plugin

Fix #80.

We won't generate a new IKFast plugin, but we should not support/release
it anymore as it was generated for a different version of the IRB 2400
URDF and thus can't work with the current robot support package.
closed 2025-04-05 by gavanderhoorn