ros-industrial/abb #94: How to disable motion cancel behavior?

Kazfyx, 2016-01-21:
In this page: http://wiki.ros.org/abb/Tutorials/RunServer , it said that ROS may try to cancel the move, when it takes longer than expected. When I use ROS-I controlling an ABB robot, this situation often occurs. How to disable this behavior?

gavanderhoorn, 2016-01-21:
This would've been a nice question for ROS Answers, so I'll link to an earlier one: [How do I disable execution_duration_monitoring ?](http://answers.ros.org/question/196586/how-do-i-disable-execution_duration_monitoring/). The question is about Fanuc, but that system is completely driver agnostic, so applies to all ROS(-I) moveit configs.

Alternatively you can disable the monitoring at runtime using `rqt_reconfigure`.

Kazfyx, 2016-01-21:
@gavanderhoorn, Thank you! It works now.

closed 2016-02-04 by Levi-Armstrong