ros-industrial/abb #96: Support for ABB IRB 120?

SteveMacenski, 2016-01-25:
Hi!

I'm starting to do work with an ABB IRB 120 and wondering if there was support for this via moveit_configs or URDF/XACRO files to jump off from. 

Thanks!

gavanderhoorn, 2016-01-25:
Please see the packages in [abb_experimental](https://github.com/ros-industrial/abb_experimental).

SteveMacenski, 2016-01-25:
Awesome, thanks! I hadn't seen that before. I appreciate the help!

closed 2016-01-25 by SteveMacenski

SteveMacenski, 2016-02-09:
For the ABB_experimental, should I grab the abb_driver node from this repo as the abb_experimental software does not include robot hardware drivers?

gavanderhoorn, 2016-02-09:
On 9-2-2016 5:37, Steven Macenski wrote:
> For the ABB_experimental, should I grab the abb_driver node from this repo as the abb_experimental software does not include robot hardware drivers?

Yes. Experimental repositories provide /additional/ packages, that are
waiting to be migrated to the non-experimental repositories. For
everything else you should just use either released or non-experimental
packages.

PS: a `rosdep check --from-paths /path/to/your/catkin_ws/src
--ignore-src` would've suggested this as well.

SteveMacenski, 2016-02-09:
Thanks! Is there some documentation or discussion as to how to use this ROS system with the hardware? (computer to controller/robot interface, sp. launch files for hardware, etc) If one doesn't exist, I would like to assist in making one. I think that would greatly help the new ABB and ROS users.

gavanderhoorn, 2016-02-09:
See [wiki.ros.org/abb](http://wiki.ros.org/abb).

gavanderhoorn, 2016-02-09:
re: 'launch files for hardware', see [WorkingWithRosIndustrialRobotSupportPackages](http://wiki.ros.org/Industrial/Tutorials/WorkingWithRosIndustrialRobotSupportPackages).

For general ROS questions: [answers.ros.org](http://answers.ros.org/questions/).

We always welcome (constructive) criticism, so if you feel something isn't correct, or missing, let us know.

SteveMacenski, 2016-02-09:
Thanks for the links! My confusion is mostly when reading the tutorials from the wiki.ros.org/abb/Tutorials. Am I correct in saying those are the steps required on the controller to configure it for taking ROS commands? I'm not sitting in front of the controller right now, but then is there a USB port or something I'm supposed to plug-in to with a computer running ROS that is generating the commands? Or doe the controller include some computer I should load ROS onto and operate from that? 
Thanks

gavanderhoorn, 2016-02-10:
The documentation expects some familiarity with setting up and configuring ABB controllers, and a basic understanding of ROS. I would highly recommend reading up on both before trying any of this.

As to your questions: the tutorials shows how to install the necessary files on the controller and how to configure it. Communication between the controller and your ROS pc (which is *not* inside your controller) is done over standard TCP/IP.

Note that running the driver in RobotStudio is also supported, so you could start with that.

SteveMacenski, 2016-02-10:
I have a good amount of experience with ROS, but the ABB controllers is something new to me. From looking inside of it, there isn't a clear 'plug in auxiliary computer here', referencing user guide as soon as I can get my hands on a copy.