ros-industrial/abb #113: Ros_messages raw data for velocity

merium, 2016-12-20:
When I insert the TPWrite command in the routine ROS_receive_msg_joint_traj_pt (code shown below), from ROS_messages module, I can see the value sent from ROS side. However, message.velocity does not show anything. If I send any velocity value for a particular joint from ROS, how can I access it in RAPID?

    UnpackRawBytes raw_message.data, 1, message.sequence_id, \IntX:=DINT;
    UnpackRawBytes raw_message.data, 5, message.joints.rax_1, \Float4;
    UnpackRawBytes raw_message.data, 9, message.joints.rax_2, \Float4;
    UnpackRawBytes raw_message.data, 13, message.joints.rax_3, \Float4;
    UnpackRawBytes raw_message.data, 17, message.joints.rax_4, \Float4;
    UnpackRawBytes raw_message.data, 21, message.joints.rax_5, \Float4;
    UnpackRawBytes raw_message.data, 25, message.joints.rax_6, \Float4;
    UnpackRawBytes raw_message.data, 29, message.velocity, \Float4;
    TPWrite "value is " + ValToStr(message.joints.rax_1);
    UnpackRawBytes raw_message.data, 29+(ROS_MSG_MAX_JOINTS-6)*4, message.velocity, \Float4;
    TPWrite "Q4 value is " + ValToStr(message.velocity);

gavanderhoorn, 2017-01-20:
I'm not quite sure I understand what you're trying to do. Could you elaborate a bit?

gavanderhoorn, 2017-01-20:
Are you asking why `message.velocity` is not populated?

merium, 2017-01-23:
Yes I'm asking why message.velocity is not populated and how to do so ?

gavanderhoorn, 2017-01-23:
Did you add this line:
```
UnpackRawBytes raw_message.data, 29, message.velocity, \Float4;
```

Because I can't find it in the [current version](https://github.com/ros-industrial/abb/blob/1c49136fcba61371e26f32a5c3e92870e1443519/abb_driver/rapid/ROS_messages.sys#L102-L112).

And with 'not populated', do you mean no value at all (which would be strange), or a `0.0`?

merium, 2017-01-24:
Yes I've tried this line and it shows 0.0. I've tried replacing 29 up to 33 and all it shows is 0.0, even when I'm sending data in velocity on the ROS side.

gavanderhoorn, 2017-01-24:
Those indices are for the entries in the `joint_data` array that are unused in the `abb_driver`:

> Skip bytes 29-44. UNUSED. Reserved for Joints 7-10

They are set to `0.0` by the ROS side, so I'm not surprised that you get `0.0` after deserialisation.

As to why the last `velocity` field is `0.0`: that is set by the ROS side as well, as the `abb_driver` uses the `duration` field. See [abb_driver/src/abb_joint_downloader_node.cpp, lines 68 to 73](https://github.com/ros-industrial/abb/blob/1c49136fcba61371e26f32a5c3e92870e1443519/abb_driver/src/abb_joint_downloader_node.cpp#L68-L73) and [abb_driver/rapid/ROS_motion.mod, line 65](https://github.com/ros-industrial/abb/blob/1c49136fcba61371e26f32a5c3e92870e1443519/abb_driver/rapid/ROS_motion.mod#L65).

merium, 2017-01-26:
I see, so I'll have to modify code on both sides to be able to send velocity. Thanks!

gavanderhoorn, 2017-01-26:
> I see, so I'll have to modify code on both sides to be able to send velocity

No, just the ROS side, and specifically those lines that I linked you to. The RAPID code already deserialises the field, but it's just set to `0.0` right now.

Note btw that the `duration` field is essentially somewhat of the inverse of `velocity` field, and also that `velocity` is not a joint velocity in rad/s or deg/s, but is supposed to be a ratio encoding the max velocity for the controller to attain over the associated traj segment (see also [Message Structures of the ROS-Industrial Simple Message Protocol - Message Definitions - JOINT_TRAJ_PT](https://github.com/gavanderhoorn/rep-ros-i/blob/344c0f56673bd78af7c2523e0e023b42982c06c9/rep-ixxxx.rst#joint-traj-pt)).

gavanderhoorn, 2017-01-26:
As I have the impression that your question has been answered, I'm going to close this issue.

If you feel that is not correct, feel free to re-open it.

closed 2017-01-26 by gavanderhoorn