ros-industrial/abb #133: catkin_make error

xsmxsm, 2017-05-24:
When I cmake this package, it occur this error(I use ubuntu 16.04 , ROS kinetic):

[ 92%] Building CXX object industrial_core-kinetic-devel/industrial_robot_client/CMakeFiles/joint_trajectory_action.dir/src/generic_joint_trajectory_action_node.cpp.o
CMakeFiles/motion_download_interface.dir/src/generic_joint_downloader_node.cpp.o：在函数‘main’中：
generic_joint_downloader_node.cpp:(.text+0x70)：对‘ros::init(int&, char**, std::string const&, unsigned int)’未定义的引用
CMakeFiles/motion_download_interface.dir/src/generic_joint_downloader_node.cpp.o：在函数‘industrial_robot_client::joint_trajectory_interface::JointTrajectoryInterface::JointTrajectoryInterface()’中：
generic_joint_downloader_node.cpp:(.text._ZN23industrial_robot_client26joint_trajectory_interface24JointTrajectoryInterfaceC2Ev[_ZN23industrial_robot_client26joint_trajectory_interface24JointTrajectoryInterfaceC5Ev]+0x69)：对‘ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_utils.so：对‘urdf::Model::initParam(std::string const&)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_robot_client.so：对‘ros::param::get(std::string const&, int&)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_robot_client.so：对‘ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_robot_client.so：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_robot_client.so：对‘ros::param::has(std::string const&)’未定义的引用
/home/xsm/catkin_wsCMakeFiles/robot_state.dir//devel/lib/libindustrial_robot_client.so：对‘ros::param::get(std::srcstring /const&generic_robot_state_node.cpp.o：�, ���std:���:string��&)��’朘main��‮�义��的引用
��/home��/xsm/catkin_ws/�
devel/lib/libindustrial_utils.so：对‘ros::param::getgeneric_robot_state_node.cpp(:(.std::textstring+0x70 const)�&,�� XmlRpc��::XmlRpcValue&)��’最��定��ros�::��init(��int&��, 用
char**,collect2: error: ld returned 1 exit status
 std::string const&, unsigned int)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_utils.so：对‘urdf::Model::initParam(std::string const&industrial_core-kinetic-devel/industrial_robot_client/CMakeFiles/motion_download_interface.dir/build.make:124: recipe for target '/home/xsm/catkin_ws/devel/lib/industrial_robot_client/motion_download_interface' failed
)make[2]: *** [/home/xsm/catkin_ws/devel/lib/industrial_robot_client/motion_download_interface] Error 1
’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_robot_client.so：对‘ros::param::get(std::string const&, int&)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_robot_client.so：对‘ros::console::print(CMakeFiles/Makefile2:7180: recipe for target 'industrial_core-kinetic-devel/industrial_robot_client/CMakeFiles/motion_download_interface.dir/all' failed
make[1]: *** [industrial_core-kinetic-devel/industrial_robot_client/CMakeFiles/motion_download_interface.dir/all] Error 2
ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocatormake[1]: *** 正在等待未完成的任务....
<char> > const&, char const*, int, char const*)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_robot_client.so：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_robot_client.so：对‘ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_robot_client.so：对‘ros::param::has(std::string const&)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_robot_client.so：对‘ros::param::get(std::string const&, std::string&)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_utils.so：对‘ros::param::get(std::string const&, XmlRpc::XmlRpcValue&)’未定义的引用
collect2: error: ld returned 1 exit status
industrial_core-kinetic-devel/industrial_robot_client/CMakeFiles/robot_state.dir/build.make:124: recipe for target '/home/xsm/catkin_ws/devel/lib/industrial_robot_client/robot_state' failed
make[2]: *** [/home/xsm/catkin_ws/devel/lib/industrial_robot_client/robot_state] Error 1
CMakeFiles/Makefile2:8121: recipe for target 'industrial_core-kinetic-devel/industrial_robot_client/CMakeFiles/robot_state.dir/all' failed
make[1]: *** [industrial_core-kinetic-devel/industrial_robot_client/CMakeFiles/robot_state.dir/all] Error 2
[ 92%] Building CXX object industrial_core-kinetic-devel/industrial_robot_client/CMakeFiles/joint_trajectory_action.dir/src/joint_trajectory_action.cpp.o
[ 93%] Linking CXX executable /home/xsm/catkin_ws/devel/lib/industrial_robot_client/motion_streaming_interface
CMakeFiles/motion_streaming_interface.dir/src/generic_joint_streamer_node.cpp.o：在函数‘main’中：
generic_joint_streamer_node.cpp:(.text+0x70)：对‘ros::init(int&, char**, std::string const&, unsigned int)’未定义的引用
CMakeFiles/motion_streaming_interface.dir/src/generic_joint_streamer_node.cpp.o：在函数‘industrial_robot_client::joint_trajectory_interface::JointTrajectoryInterface::JointTrajectoryInterface()’中：
generic_joint_streamer_node.cpp:(.text._ZN23industrial_robot_client26joint_trajectory_interface24JointTrajectoryInterfaceC2Ev[_ZN23industrial_robot_client26joint_trajectory_interface24JointTrajectoryInterfaceC5Ev]+0x69)：对‘ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_utils.so：对‘urdf::Model::initParam(std::string const&)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_robot_client.so：对‘ros::param::get(std::string const&, int&)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_robot_client.so：对‘ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_robot_client.so：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_robot_client.so：对‘ros::param::has(std::string const&)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_robot_client.so：对‘ros::param::get(std::string const&, std::string&)’未定义的引用
/home/xsm/catkin_ws/devel/lib/libindustrial_utils.so：对‘ros::param::get(std::string const&, XmlRpc::XmlRpcValue&)’未定义的引用
collect2: error: ld returned 1 exit status
industrial_core-kinetic-devel/industrial_robot_client/CMakeFiles/motion_streaming_interface.dir/build.make:124: recipe for target '/home/xsm/catkin_ws/devel/lib/industrial_robot_client/motion_streaming_interface' failed
make[2]: *** [/home/xsm/catkin_ws/devel/lib/industrial_robot_client/motion_streaming_interface] Error 1
CMakeFiles/Makefile2:7220: recipe for target 'industrial_core-kinetic-devel/industrial_robot_client/CMakeFiles/motion_streaming_interface.dir/all' failed
make[1]: *** [industrial_core-kinetic-devel/industrial_robot_client/CMakeFiles/motion_streaming_interface.dir/all] Error 2
[ 94%] Linking CXX executable /home/xsm/catkin_ws/devel/lib/industrial_robot_client/joint_trajectory_action
CMakeFiles/joint_trajectory_action.dir/src/generic_joint_trajectory_action_node.cpp.o：在函数‘main’中：
generic_joint_trajectory_action_node.cpp:(.text+0x70)：对‘ros::init(int&, char**, std::string const&, unsigned int)’未定义的引用
CMakeFiles/joint_trajectory_action.dir/src/joint_trajectory_action.cpp.o：在函数‘industrial_robot_client::joint_trajectory_action::JointTrajectoryAction::JointTrajectoryAction()’中：
joint_trajectory_action.cpp:(.text+0x64)：对‘ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)’未定义的引用
joint_trajectory_action.cpp:(.text+0x39c)：对‘ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)’未定义的引用
[... cut ...]
industrial_core-kinetic-devel/industrial_robot_client/CMakeFiles/joint_trajectory_action.dir/build.make:169: recipe for target '/home/xsm/catkin_ws/devel/lib/industrial_robot_client/joint_trajectory_action' failed
make[2]: *** [/home/xsm/catkin_ws/devel/lib/industrial_robot_client/joint_trajectory_action] Error 1
CMakeFiles/Makefile2:7329: recipe for target 'industrial_core-kinetic-devel/industrial_robot_client/CMakeFiles/joint_trajectory_action.dir/all' failed
make[1]: *** [industrial_core-kinetic-devel/industrial_robot_client/CMakeFiles/joint_trajectory_action.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

gavanderhoorn, 2017-05-24:
I don't think we can help you with all the compiler errors in Chinese.

Could you please switch to English and update your original post?

AustinDeric, 2017-05-24:
you should use `catkin build`, not `cmake`.  This package currently builds of your system is setup correctly:
https://travis-ci.org/ros-industrial/abb

xsmxsm, 2017-05-25:
@gavanderhoorn 
The Chinese error is "Undefined reference" .
Why I can not install abb_driver through "sudo apt-get install ros-kinetic-abb-driver" ?
It can not find this package. But in indigo version , I can install with "sudo apt-get install ros-indigo-abb-driver".

xsmxsm, 2017-05-25:
CMake Warning (dev) at /opt/ros/kinetic/share/catkin/cmake/test/tests.cmake:137 (add_dependencies):
  Policy CMP0046 is not set: Error on non-existent dependency in
  add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  The dependency target "joint_trajectory_action" of target
  "run_tests_abb_irb4400_support_roslaunch-check_tests_roslaunch_test.xml"
  does not exist.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/roslaunch/cmake/roslaunch-extras.cmake:58 (catkin_run_tests_target)
  abb/abb_irb4400_support/CMakeLists.txt:10 (roslaunch_add_file_check)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /opt/ros/kinetic/share/catkin/cmake/test/tests.cmake:149 (add_dependencies):
  Policy CMP0046 is not set: Error on non-existent dependency in
  add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  The dependency target "joint_trajectory_action" of target
  "_run_tests_abb_irb4400_support_roslaunch-check_tests_roslaunch_test.xml"
  does not exist.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/roslaunch/cmake/roslaunch-extras.cmake:58 (catkin_run_tests_target)
  abb/abb_irb4400_support/CMakeLists.txt:10 (roslaunch_add_file_check)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /home/xsm/123/build

Levi-Armstrong, 2017-05-25:
@AustinDeric What is the plan on the kinetic release?

gavanderhoorn, 2017-05-25:
@xsmxsm wrote:
> Why I can not install abb_driver through "sudo apt-get install ros-kinetic-abb-driver" ?
It can not find this package. But in indigo version , I can install with "sudo apt-get install ros-indigo-abb-driver".

Because these packages haven't been released into Kinetic yet.

AustinDeric, 2017-05-27:
The release process is complete, sorry for the delay:

https://github.com/ros/rosdistro/pull/15089

AustinDeric, 2017-06-14:
@xsmxsm the binaries have been released on Ubuntu.  I recommend we close this issue.  Thank you everyone for your assistance with this issue.

closed 2017-06-15 by AustinDeric