ros-industrial/abb #134: Trajectory planning error

MilanMichael, 2017-06-26:
In ros i make a trajectory out of many waypoints and when i execute the trajectory points , The robot model in rviz is executing but in the robot studio the robot move a little and stops with an error too many trajectory points ( trajectory points is reached maximum limit which is 100).
Is there any way to solve the problem?

Levi-Armstrong, 2017-06-26:
You should be able to change this [variable](https://github.com/ros-industrial/abb/blob/kinetic-devel/abb_driver/rapid/ROS_common.sys#L36). I am not sure why it was set to 100 versus something else so you may run into other issues if you increase it to high.

MilanMichael, 2017-07-09:
Sir,
I done it , now that problem is solved. Now the robot studio receive 173 trajectory points. but the robot is not moving continuously. It execute a part of trajectory and wait, after some time the next part is executed. The whole trajectory is executed stepwise.

Levi-Armstrong, 2017-07-10:
I would check the time values of the trajectory and see if they look correct.

Levi-Armstrong, 2017-07-10:
Also [this](http://wiki.ros.org/Industrial/Tutorials/Improving_Motion_on_an_Industrial_Robot) has useful information on what you are experiencing.

gavanderhoorn, 2018-11-02:
Closing due to inactivity.

Feel free to keep commenting on the issue though.

closed 2018-11-02 by gavanderhoorn

ZhangPeike, 2019-03-07:
@Levi-Armstrong I caught the same problem,  but when trying to change the MAX_TRAJ_LENGTH in the ROS_common.sys in the driver,  RobotStudio raised an error RAPID error.