ros-industrial/abb #136: Ability to use System Inputs and System Outputs to reset e

srsidd, 2017-10-24:
When an eStop is activated, for whatever reason, it is required to 'Acknowledge' the error on the FlexPendant and turn the motors back by pressing the blinking motors on switch to continue. I need to do this with software.

I found out that we can do this with robot studio using the System Inputs and System Output. We can input `Reset Emergency Stop` until output the `Run Chain OK` and input `Motors On` until output `Motors On State` to do it with robot studio and rapid.
Is there capability to do this on ROS?

gavanderhoorn, 2017-10-24:
Not a direct answer, but a related Q&A on ROS Answers: [How to get and handle ros-industrial errors in a generic way ?](https://answers.ros.org/question/272869).

gavanderhoorn, 2017-10-24:
And the related PR: https://github.com/ThomasTimm/ur_modern_driver/pull/143.

srsidd, 2017-10-25:
Correct me if I'm wrong - that is pointing me to how we can use the `industrial_msgs::RobotStatus` message as a system monitor, not necessarily to set\reset the system inputs and outputs.

I remember that the UR robots publish a message,  `ur_msgs::IOStates`  to monitor the states of the system, and uses a service to set\reset the IO. Does the ABB ros driver have this capability too?

gavanderhoorn, 2017-10-25:
@srsidd wrote:
> Correct me if I'm wrong - that is pointing me to how we can use the `industrial_msgs::RobotStatus` message as a system monitor, not necessarily to set\reset the system inputs and outputs.

As I wrote above, it was not an answer, but connects your inquiry to the wider discussion about controller status, robot IO and interaction with the safety/error systems present on robot controllers.

gavanderhoorn, 2017-10-25:
@srsidd wrote:
> I remember that the UR robots publish a message,  `ur_msgs::IOStates`  to monitor the states of the system, and uses a service to set\reset the IO. Does the ABB ros driver have this capability too?

No, there is no IO support in `abb_driver` at this time.

Rationale for this is that it would most likely not make much sense for us (ie: ROS-Industrial) to design, create and maintain yet another fieldbus-like system when there are already so many good ones available. Those systems also come with tooling, training, hw and support from their mfgs, and integrate with industrial robot hardware in a clean and supported way.

If you are looking for a software-only solution, and your controller software is recent enough, using the RWS might be an option. This will be supported in the newer EGM based drivers that will become available in the near future, but would require some custom development if you can't wait for that.

The RWS is a HTTP based RPC-like system, so should not be too hard to interface with using something like `curl` (C/C++) or any of the Python `requests` like libraries.

gavanderhoorn, 2017-10-25:
And to tie it to the ROS Answers question and PR mentioned earlier: an extension of the relevant ROS-I driver spec that includes fault introspection and reset interaction capabilities could then wrap an RWS backed implementation to provide an abstracted and common interface to that functionality across all supported robot drivers.

rajmohan747, 2018-07-31:
since i came across the same issue....i am posting the question again...

I remember that the UR robots publish a message, ur_msgs::IOStates to monitor the states of the system, and uses a service to set\reset the IO. Does the ABB ros driver have this capability too?????

gavanderhoorn, 2018-07-31:
No, it doesn't, and will most likely also not be extended with such functionality.

I'd recommend taking a look at [ros-industrial/abb_librws](https://github.com/ros-industrial/abb_librws), which is -- if you don't want to use a fieldbus -- the recommended way to do this on recent ABB controllers.

gavanderhoorn, 2019-09-17:
With the release of `abb_librws`, there is now a way to do this, at least on controllers with RW6.

Other robots could use any fieldbus that is supported by IRC5 or their respective controller.

Extending `abb_driver` with support for what is described in the OP is not planned, but PRs adding it will also not be ignored.

closed 2019-09-17 by gavanderhoorn