ros-industrial/abb #141: Unable to connect to the robot

srsidd, 2018-01-23:
I am unable to get the `abb_driver` to communicate with the robot. I run `roslaunch abb_irb1600_support robot_interface_download_irb1600.launch robot_ip:=<ip_address>` but I get the following errors - 
```
[ERROR] [1516661727.384124332]: Failed to connect to server, rc: -1. Error: 'Connection refused' (errno: 111)
[ERROR] [1516661727.384219436]: Failed to receive message length
[ERROR] [1516661727.384249799]: Failed to receive incoming message
```
At the same time the flex pendant shows me these messages -
```
ROS_StateServer->StateServer: Waiting for connection.
ROS_MotionServer->MotionServer: Waiting for connection.
```

I'm able to ping the robot, so I know it's on the network. I tried `nc <ip_address> 11002\11000`, but doesn't output anything to the terminal.

I tried doing some debugging, and the `motion_download_interface` executes fine without any errors. The `robot_state` node, however initializes all the parameters and gets stuck at https://github.com/ros-industrial/abb/blob/kinetic-devel/abb_driver/src/abb_robot_state_node.cpp#L75.

I'm kind of lost and don't know how to proceed, does someone have any pointers?

gavanderhoorn, 2018-01-23:
@srsidd wrote:
>  I run `roslaunch abb_irb1600_support robot_interface_download_irb1600.launch robot_ip:=<ip_address>` [..]

I don't believe we have a `robot_interface_download_irb1600.launch` in `abb_irb1600_support`. Is that a file you created yourself?

> I tried `nc <ip_address> 11002\11000`, but doesn't output anything to the terminal.

that would seem to indicate that *something* is blocking you from connecting to the robot. That same *something* is probably why the nodes can't connect. `nc <ip_address> 11002` should start printing to the console as soon as you connect.

Is this a real robot, or a RobotStudio session? If the latter: have you checked the Windows firewall settings?

gavanderhoorn, 2018-01-23:
Also: please always tell us how you installed the packages, what version of ROS you're using, how you installed that, etc.

Without such information we can't be of much help.

srsidd, 2018-01-23:
We are using `robot_interface_download_irb1600x145.launch` from the [private repository ](https://github.com/ros-industrial-consortium/abb_experimental) `abb_experimental` in the `ros-industrial-consortium` which is probably why that launch file is not visible here.

The system is Ubuntu 16.04, with `ros-kinetic` installed using the binary. The `abb_experimental` was cloned from source.

This is the real robot, not robot studio.

gavanderhoorn, 2018-01-24:
> We are using `robot_interface_download_irb1600x145.launch` from the [private repository ](https://github.com/ros-industrial-consortium/abb_experimental) `abb_experimental` in the `ros-industrial-consortium` which is probably why that launch file is not visible here.

ok. May I suggest:

 1. use only the EGM driver packages from the `ros-industrial-consortium/abb_experimental` repository, nothing else
 1. when using `abb_driver`, use only packages from `ros-industrial/abb` and `ros-industrial/abb_experimental`

I know this is less than ideal right now, but mixing this is going to confuse people (me and you).

At this point the biggest clue we have is that `nc` also can't seem to connect.

Can you check with Wireshark whether it can actually connect, but just doesn't get any data, or doesn't actually connect?

Also: make sure there are no wireless segments between you and the robot controller, *anywhere*.

gavanderhoorn, 2018-01-24:
@gavanderhoorn wrote:
> 1 . use only the EGM driver packages from the `ros-industrial-consortium/abb_experimental` repository, nothing else

This is the same advice I gave you in https://github.com/ros-industrial-consortium/abb_experimental/issues/11.

srsidd, 2018-01-30:
Took your advice, and tried the package from the publicly available driver. Ran into the same problem. We realized that we were not able to resolve the ip address from the rapid side. Specifically, `GetSysInfo(\LanIp)` [here](https://github.com/ros-industrial/abb/blob/kinetic-devel/abb_driver/rapid/ROS_socket.sys#L32) returned empty. We were able to connect the robot by hard coding the ip address in it's place.
Not sure what the real problem is.

gavanderhoorn, 2018-01-31:
> Took your advice, and tried the package from the publicly available driver. Ran into the same problem. 

a bit captain hindsight, but this was something to be expected.

> We realized that we were not able to resolve the ip address from the rapid side. Specifically, `GetSysInfo(\LanIp)` [...]

That is strange, and would be something to look into.

Does this happen with any RAPID program (ie: with just a `GetSysInfo(\LanIp)` piped to a popup or print statement fi)?

gavanderhoorn, 2018-02-14:
@srsidd: did you figure out what was going on with `GetSysInfo(\LanIp)`?

srsidd, 2018-02-15:
Sorry for the late reply, I tried your suggestion - the `GetSysInfo(\LanIp)` returned WanIp and not the LanIp. But according to the [manual](http://developercenter.robotstudio.com/BlobProxy/manuals/RapidIFDTechRefManual/doc378.html) it is supposed to return the WanIp. 
We were able to connect to the robot when we hard coded the LanIp which makes sense... Not sure what the solution is though.

gavanderhoorn, 2018-02-15:
Which network connection are you using? The service port, or the regular one?

srsidd, 2018-02-15:
We're using the regular one - the X5. LAN3.

gavanderhoorn, 2018-02-15:
According to [this page](http://developercenter.robotstudio.com/BlobProxy/manuals/SysParametersTechRefManual/doc7.html), `X3` is the connection that the flexpendant should be connected to, part of the *private network*. Shouldn't `X6` be used for regular connections?

srsidd, 2018-02-15:
Ah, good catch. I don't remember why we had it set up that way. Changing that to X6 which is the `WAN` network should work. Thanks! Closing this issue.

closed 2018-02-15 by srsidd

gavanderhoorn, 2018-02-16:
It shouldn't, but I'd be interested to know whether #142 is also influenced by this.

nali12, 2019-02-20:
I am having this same problem. but I am working with RobotStudio. help please.

gavanderhoorn, 2019-02-20:
@nali12 wrote:
> I am having this same problem. but I am working with RobotStudio.

Without further information it's hard to know whether you are experiencing the same problem (and given the context of the OP I'm not entirely sure you are).

Please post a new issue about your problem and include more information: OS, ROS (version, how installed), which driver component you are using, how you installed all of that (on the ROS side, on the robot side), how you setup your (virtual controller), etc, etc.

nali12, 2019-02-20:
So, I am unable to make the `abb_driver`to communicate with my robot in RobotStudio. I am running the following command: `roslaunch abb_irb120_support robot_interface_download_irb120_3_58.launch robot_ip:=<ip_address>` , but the only error I get are the following:
```
[ERROR] [1550679332.164635951]: Failed to connect to server, rc: -1. Error: 'Connection refused' (errno: 111)
[ERROR] [1550679332.164717488]: Failed to receive message length
[ERROR] [1550679332.164736088]: Failed to receive incoming 
```
and the VC just keep showing the same message

```
ROS_StateServer->StateServer: Waiting for connection.
ROS_MotionServer->MotionServer: Waiting for connection.
```
I am using `robot_interface_download_irb120_3_58.launch` from the repostory `abb_experimental` (http://wiki.ros.org/abb_experimental). My system is Ubuntu 16.04, with `ROS kinetic`. In order to control the robot in RobotStudio I installed the ABB ROS server following all the tutorials (http://wiki.ros.org/abb/Tutorials).

gavanderhoorn, 2019-02-20:
Please post a new *issue*, not a new comment ..