ros-industrial/abb #148: Client is not connecting

arjunskumar, 2018-04-03:
I am running Ubuntu 16.04 with Kinetic Kame. And I have a Windows10 PC with RobotStudio 6.0 6 running. The IPv4 address in the Windows machine is 192.168.43.1

I followed the tutorials as in ROS Wiki
1. http://wiki.ros.org/abb/Tutorials/RobotStudio
2. http://wiki.ros.org/abb/Tutorials/InstallServer
3. http://wiki.ros.org/abb/Tutorials/RunServer

so I modified the lines in GetSysInfo() patch with 192.168.43.1 as the IP address like 

IF (SocketGetStatus(server_socket) = SOCKET_CREATED) SocketBind server_socket, "192.168.43.1", port;

Now, I launch RobotStudio and virtual flex pendant. I put the controller in Auto mode and start the program

On the flexpendant, I have this showing on the flexpendant: 
ROS_StateServer->StateServer: Waiting for connection.
ROS_MotionServer->MotionServer: Waiting for connection.

On the ROSPC, I tried roslaunch abb_irb1200_support robot_interface_download_irb1600.launch robot_ip:=192.168.43.1, 
and getting rostopics
rostopic list
/feedback_states
/joint_path_command
/joint_trajectory_action/cancel
/joint_trajectory_action/feedback
/joint_trajectory_action/goal
/joint_trajectory_action/result
/joint_trajectory_action/status
/robot_status
/rosout
/rosout_agg

But flex pendant is not showing connectec to client..

gavanderhoorn, 2018-04-03:
Please make sure the Windows firewall is not interfering. There are separate settings for 'private' and 'public' networks.

As a first test, disable the firewall completely (make sure you're not connected to any networks with world-routable IPs, just to be sure) and try to connect. If that works, you'll need to update your firewall configuration.

gavanderhoorn, 2018-04-03:
And I'm assuming that your ROS PC can actually communicate with the PC running RobotStudio here (ie: can `ping`).

To get some more info, try to start the launch files with:

```
roslaunch --screen abb_irb1200_support robot_interface_download_irb1600.launch robot_ip:=192.168.43.1
```

arjunskumar, 2018-04-03:
Yes I disabled the firewall, and able to ping the ip address. 

I also have a real  ABB robot, but still has the same problem.. Client is not working, I dont know where I'm wrong.

gavanderhoorn, 2018-04-03:
Ok. With the programs running on the robot, what is the output on the ROS PC if you run this in a terminal:

```
nc 192.168.43.1 11002
```

gavanderhoorn, 2018-04-03:
I only now noticed this:

> But flex pendant is not showing connectec to client..

afaik there is no visual confirmation of established connections on the TP side.

If you start the driver, does a `rostopic echo /joint_states` output expected data? If it does, the driver would seem to be working.

arjunskumar, 2018-04-05:
Hi, Finally it worked, I connect ROS-PC with Service Port and fired the launch files.
Now I'm able to work both in Real and Simulated Robot
Thanks Much

closed 2018-04-05 by arjunskumar

gavanderhoorn, 2018-04-05:
So to recap:

 1. it worked already with RobotStudio?
 1. you're now using the service port on the real robot and that also makes it work?

It's up to you, but afaik ABB does not recommend using the service port for regular LAN traffic. There are dedicated ports inside the controller for that.

Also note that the changes to the `SocketBind server_socket, ..` line should only be made for RobotStudio, not for the real robot.

arjunskumar, 2018-04-05:
1) Yes worked perfectly with RobotStudio
2) It worked using service port on the real robot

Thanks for the information, I have changed to WAN port, now it is working.

Thanks Much