ros-industrial/abb #149: How to control the velocity of the end effector?

EdXian, 2018-04-14:
Hi all,
I have to control the velocity and acceleration of the end effector due to research purpose .
Is it posible to control these states in simulation?

gavanderhoorn, 2018-04-14:
Simulation: yes, but with some work. It should be possible to use `gazebo_ros_control` and to write a Cartesian controller that will allow you to do what you want. I'm not aware of anything immediately reusable.

Real robot: velocity: not with `abb_driver`, maybe with [abb_libegm](https://github.com/ros-industrial/abb_libegm). Acceleration (effort?): I'm not aware of a way to control that directly with ABB robots.

gavanderhoorn, 2018-04-14:
Could I ask you to please ask questions like this on [ROS Answers](http://answers.ros.org)? This issue tracker is for reporting problems with the packages in this repository. Your question is more a general inquiry.

closed 2018-04-14 by gavanderhoorn

EdXian, 2018-04-14:
Hi @gavanderhoorn,
Ok , thanks you for your information.