ros-industrial/abb #156: Problem working with real IRB 1200 robot

rajmohan747, 2018-11-16:
Hi,
I am with IRB 1200 in ROS-Kinetic platform.
Initially I am launching 

```
roslaunch abb_irb1200_support robot_interface_streaming_irb1200_5_90.launch robot_ip:=192.168.125
```

Since I'm working with real robot, _sim_ parameter was set as `True`. On launching ,

```
roslaunch abb_irb1200_5_90_moveit_config moveit_planning_execution.launch sim:=false robot_ip:=192.168.125.1
```

I am facing an error as `robot_interface_streaming_irb1200_5_90.launch` is not available.

Also tried replacing `robot_interface_streaming_irb1200_5_90.launch` with `robot_interface_download_irb1200_5_90.launch`.

After that on launching the node to move the robot,the teach pendant is showing like:

```
Sending  24 points to MOTION Task.  Traj START received Traj END received
```

But the robot is still not moving

Please correct me, if I'm doing something wrong

gavanderhoorn, 2018-11-17:
Please see whether #155 fixes this.

gavanderhoorn, 2018-12-06:
I'm going to close this as I believe #155 fixes what @rajmohan747 described.

If that is not the case then please re-open and clarify.

closed 2018-12-06 by gavanderhoorn