ros-industrial/abb #157: Where to edit joint state messages?

krebsalad, 2018-12-03:
I am not sure if this is the right place to ask this. 

I want to edit the messages sent to the controller, namely the joint angles. I want to add a certain offset to each joint angle based on the calibration tag in the robots URDF. I am assuming that this is currently not a feature in the driver, at least, I could not find something regarding this in the abb_driver nor in the industrial_robot_client. Can someone point me to where I can best add these offsets to the joint angles/joint states?

somewhere here? https://github.com/ros-industrial/industrial_core/blob/kinetic-devel/industrial_robot_client/src/joint_relay_handler.cpp#98

gavanderhoorn, 2018-12-03:
Context: https://github.com/mikeferguson/robot_calibration/issues/64.

gavanderhoorn, 2018-12-03:
Adding support for calibration offsets could be interesting.

> somewhere here? https://github.com/ros-industrial/industrial_core/blob/kinetic-devel/industrial_robot_client/src/joint_relay_handler.cpp#98

That doesn't link me to a specific line number ([this](https://github.com/ros-industrial/industrial_core/blob/kinetic-devel/industrial_robot_client/src/joint_relay_handler.cpp#L98) does) so I assume you are referring to `JointRelayHandler::create_messages(..)`.

There are two sides to this:

 1. messages coming from the driver server
 1. messages going to the server

The first needs a transformation from `SimpleMessage` to the appropriate ROS message, while the second needs the inverse transformation.

For calibration to be taken into account properly (and I'm assuming you're talking about joint space offsets), you'd have to subtract the offsets in the first case, and add them in the second.

The bigger issue here I believe would be to make the offsets themselves available to the conversion functions without introducing all sorts of extra infrastructure.

gavanderhoorn, 2018-12-03:
It would be nice if we could add this to `industrial_robot_client` itself, as that would add support for this to all robot drivers making use of that.

In this specific case though (the `abb_driver`), it would probably have to be added to `ABB_JointRelayHandler` as well, as that subclasses the regular `JointRelayHandler`. But whether that is really needed will depend on how things will be implemented exactly.

krebsalad, 2018-12-03:
> I assume you are referring to JointRelayHandler::create_messages(..)

Yes, I was referring to that function. Thanks for your confirmation.

> For calibration to be taken into account properly (and I'm assuming you're talking about joint space offsets), you'd have to subtract the offsets in the first case, and add them in the second.

I will try implementing this in my project. I will report back here in the coming week.

gavanderhoorn, 2018-12-03:
> Or did i understand it the wrong way?

yes, that is not what I meant.

The function you link to only works in one direction. You need to update the functions for *both* directions of data flow.

> Also, should I change the title to something in the lines of: JointRelayHandler calibration offsets support?

yes, that would seem like a good thing to do.

I might even transfer this issue to the `ros-industrial/industrial_core` issue tracker, as it's not something specific to the `abb_driver` implementation.