ros-industrial/abb #158: ABB IRB120 does not move when trajectory command is sent from Moveit

jclinton830, 2018-12-12:
Hi, 

I followed this [tutorial](http://wiki.ros.org/abb/Tutorials/InstallServer) to configure ROS and an ABB IRB120 robot to work together. 

I then launched this launch file:
`roslaunch abb_irb120_support robot_interface_download_irb120_3_58.launch robot_ip:=192.168.125.1`

and then this launch file:
`roslaunch abb_irb2400_moveit_config moveit_planning_execution.launch robot_ip:=192.168.125.1`

Then I am able to mirror the movement of the actual robot on RViz accurately when the robot moves during jogging or running existing rapid programs in the controller. 

But when planning a motion using Rviz and sending it to the robot controller, it says that Sending 20 points to MOTION task

But the robot does not move at all. 

Anyone know why this might be happening? I am using ROS Kinetic and working on ubuntu 16.04.

Thanks in advance!

gavanderhoorn, 2018-12-12:
Three things to check:

 1. you've made sure to start the motion task (also required in Auto mode)
 1. you're using current `kinetic-devel` of `abb_driver` (specifically: with #155
 1. you're starting a motion where the first trajectory point coincides with the current state of the robot (if you don't, there is a very long duration (10 sec) slow motion included which you may see as the robot "not moving"). See also #142 and #146.

jclinton830, 2018-12-12:
HI @gavanderhoorn,

Thank you for the quick reply. 
I was trying the motion execution on manual mode. I have already tried it in Auto mode as well and it doesn't work either.

In this [tutorial](http://wiki.ros.org/abb/Tutorials/InstallServer) it is mentioned that 

> 
If T_ROB1 has existing motion-control modules, you may need to rename the main() routine in ROS_Motion.mod to ROS_main(). In this case, set the Entry point for T_ROB1 task to ROS_main().
> 

Is this a compulsory change? When I tried this it did not work either. 

I am guessing its to do it with how I run the ROS server from the teach pendant. When I go in to the production window and press PP to main, and enable the motors and press play the robot starts to execute an old piece of rapid code stored in the controller.

gavanderhoorn, 2018-12-12:
You'll have to make sure that the driver does not interact/interfere with other code present on the controller.

I cannot help you with that, as it's very specific to your setup.

gavanderhoorn, 2018-12-12:
@jclinton830: for now, I'm going to assume this is a configuration issue on your side (ie: the controller) and not something with these packages.

I'll close this issue here for now, but if you feel there is something we should change or improve then please feel free to comment here.

closed 2018-12-12 by gavanderhoorn

jclinton830, 2018-12-13:
Hi @gavanderhoorn 

I did a complete re setup of the ros workspace last night and tried again this morning on the actual robot. 

Now the robot is receiving the trajectory, but a STOP command is issued before any motion is executed. I thought #62 fixed this issue. But seems like there is still a problem. 

[image]

gavanderhoorn, 2018-12-13:
I've already commented on #41 where you also posted this (could you please not do this in the future?). From https://github.com/ros-industrial/abb/issues/41#issuecomment-446860150:

> Please see [How do I disable execution_duration_monitoring ?](https://answers.ros.org/question/196586). The robot is too slow and MoveIt cancels execution of your trajectory because of that.

gavanderhoorn, 2018-12-14:
> Thanks that fixed the problem for shorter paths but the problem persists for longer motion paths.

Then I'm guessing you changed the scale / value, but did not disable.

> Is there a parameter that can be edited within the move_group cpp source files to solve this?

No. And you should not look for that either. This is configuration work, not the "edit the source code" side of things.

jclinton830, 2018-12-14:
Did both but was still having problems, figured out it was to do with the beachnamespaces. Anyway I got it to work now. Although the motions are always executed after a delay of 20 seconds after the motion is planned. Do you know what triggers this delay? It always takes 20s to execute the trajectory, I have timed it a few times.

jclinton830, 2019-01-08:
> Can you check bullet 2 in #158 (comment)?

@gavanderhoorn Sorry about my late reply, I checked the branch name and it is kinetic-devel.

[image]

> In any case: if you're using rqt_reconfigure, you'll have to redo that every time you start everything up. Dynamic reconfiguration changes are not persisted.

Do you think this causes an issue with the delay? If so then how do I find out if rqt_reconfigure is in use?

Still having the 20 seconds of delay before the execution of a trajectory. Any help would be much appreciated. Thank you.

gavanderhoorn, 2019-01-08:
> > Can you check bullet 2 in #158 (comment)?
> 
> @gavanderhoorn Sorry about my late reply, I checked the branch name and it is kinetic-devel.

It's important that it's not just `kinetic-devel`, but actually a (very) recent checkout.

Can you make sure you have 4b11bf6b in your clone *and* installed on the robot?

jclinton830, 2019-01-09:
> Can you make sure you have 4b11bf6 in your clone and installed on the robot?

@gavanderhoorn After looking through the ROS_Motion.MOD file on both clone and the robot controller I can confirm that it matches with [4b11bf6](https://github.com/ros-industrial/abb/commit/4b11bf6bfab44e6c60318620123ded18e81a5b1a).

Any other suggestions??

gavanderhoorn, 2019-01-09:
Just to diagnose this: can you completely remove everything related to the driver from your controller and try [this version](https://github.com/ros-industrial/abb/tree/10411781b2a0f2f0d33bf3c14de19624ce757518/abb_driver/rapid)?

I don't understand the 20 seconds, as that is specified nowhere in the code (10 seconds is though).

jclinton830, 2019-01-14:
We tried this, and that did not help either :(

gavanderhoorn, 2019-01-14:
Ok.

I don't understand where the *20 seconds* come from, as that is not a nr used anywhere in the code afaict.

Can you please provide a complete description of:

 - versions of code used (ROS, `abb_driver` (`git rev-parse HEAD`) , other ABB pkgs) *including* local changes to pkgs
 - your workspace (how you created it, what pkgs are in it, etc)
 - your robot controller: RW version, any special software options
 - how you run everything, both:
   - on the ROS side (`roslaunch` commands, any RQT UIs used and how, how you command motion)
   - on the robot/controller side (buttons pressed (on both the controller *and* the TP), controller's mode, etc)