ros-industrial/abb #164: OmniCore and IRB 1100 support

garyedwards, 2019-01-11:
I may be jumping the gun here as I believe it is released this month, but we are in the experimental repo.  Has there been any testing of the new OmniCore controller with ROS?  I contacted my local ABB support who had not head of ROS but suggested it can run the same rapid code as older larger models.  We are looking at procuring a small / mobile robot at the moment and it would be great to understand if this is likely to be supported.

gavanderhoorn, 2019-01-11:
I've also transferred this to the main repository as the driver you're referring to is hosted here.

gavanderhoorn, 2019-02-20:
@garyedwards: as the new controllers haven't really been released yet, we can't really act on this right now. In addition, according to @jontje, the new controller should be fully bw-compatible, so the assumption right now would be that `abb_driver` "just works".

Based on this I'm closing this issue as I believe @jontje answered your question.

If it turns out the driver doesn't work and we need to change something -- as far as we don't all migrate to `abb_libegm` and `abb_librws` -- we can re-open.

closed 2019-02-20 by gavanderhoorn