ros-industrial/abb #165: Not able to execute cartesian trajectory on ABB IRB120

tbs845, 2019-01-15:
I am using using Ubuntu 16.04, Kinetic distro, ROS ABB experimental and ABB Kinetic-devel repos.

I am trying to get the robot to do a simple Cartesian trajectory. It is working on RViz simulation but the controller is not receiving the trajectory. Therefore the robot is not executing it. 
It works when I send individual points but not a trajectory.

You can find my node generating the Cartesian path [here](https://github.com/tbs845/cartesian_path_ROS/blob/master/threedprint_waypoints_node.cpp). 

For your reference I have attached a screen shot of the terminal of what the RViz is reporting when I ROS launch my node.
[image]

I am not sure what is going on, I am following the tutorials, new to ROS. Any guidance would be much appreciated.

tbs845

gavanderhoorn, 2019-01-15:
First: please do not post screenshots of terminal text.

It's just text, so *copy-paste* it into the issue. Use proper formatting for it then (three backticks before and after).

---

Two things to check:

 - why is `joint_trajectory_action` shutting down? Without that, nothing will move.
 - which MoveIt launch file are you starting? If it's `demo.launch`, nothing will move.

---

Finally:

> I am following the tutorials

That's great, but I hope you agree that without you telling us *which* tutorials you are following exactly, this statement is not going to help us help you.

gavanderhoorn, 2019-01-15:
And as this is most likely not an issue with this package, but a configuration/usage issue, I'm going to close this ticket.

That does not mean that we can't continue diagnosing your issue though, so feel free to keep commenting on it.

closed 2019-01-15 by gavanderhoorn