ros-industrial/abb #166: Unable to connect ROS with Robotstudio

nali12, 2019-02-21:
So, I am unable to make the `abb_driver`to communicate with my robot in RobotStudio. I am running the following command: `roslaunch abb_irb120_support robot_interface_download_irb120_3_58.launch robot_ip:=<ip_address>` , but the only error I get are the following:

```
[ERROR] [1550679332.164635951]: Failed to connect to server, rc: -1. Error: 'Connection refused' (errno: 111)
[ERROR] [1550679332.164717488]: Failed to receive message length
[ERROR] [1550679332.164736088]: Failed to receive incoming 
```
and the VC just keep showing the same message

```
ROS_StateServer->StateServer: Waiting for connection.
ROS_MotionServer->MotionServer: Waiting for connection.
```
I am using `robot_interface_download_irb120_3_58.launch` from the repository `abb_experimental` (http://wiki.ros.org/abb_experimental). My system is Ubuntu 16.04, with `ROS kinetic`, Virtual machine: Parallels. In order to control the robot in RobotStudio I installed the ABB ROS server following all the tutorials (http://wiki.ros.org/abb/Tutorials).

gavanderhoorn, 2019-02-21:
> I am running the following command: `roslaunch abb_irb120_support robot_interface_download_irb120_3_58.launch robot_ip:=<ip_address>`

Just making sure: you are specifying a real IP here, not `<ip_address>`, correct?

Also: make sure the Windows firewall is not blocking RobotStudio from accepting incoming connections.

nali12, 2019-02-21:
Yes, I am specifying the static IP that I set on my Windows Computer. Windows is the host and Ubuntu is inside Parallels. Windows computer is connected to Wifi, I don't know if that matters too. Inside ROS_socket I set the IP assigned in my windows PC: 
```
IF (SocketGetStatus(server_socket) = SOCKET_CREATED) SocketBind server_socket, "192.168.1.2", port;
```

gavanderhoorn, 2019-02-21:
Can you ping the Windows IP from the Parallels VM?

I would still try to disable the firewall, just for a test.

gavanderhoorn, 2019-02-21:
As I'm pretty sure this is a configuration issue very specific to your setup (and not a problem with the pkg in this repository), I'm going to close this issue.

If it turns out there is actually something we can fix by changing `abb_driver` or any of the other pkgs, we'll re-open.

Feel free to keep commenting on this issue though.

closed 2019-02-21 by gavanderhoorn

gonzalocasas, 2019-02-21:
You could try using `"0.0.0.0"` in `ROS_socket` to make sure it binds to all your interfaces (especially for a virtual controller, for real controllers you might want to be pickier in choosing only the right one).

And as @gavanderhoorn, make sure you can ping your Windows IP from within the virtual machine, and if that doesn't work, check the networking mode you have selected in Parallels (`Bridged` might be the easiest to work with, since the VM behaves just as another machine on the same network as the host machine's network).

nali12, 2019-02-21:
I tried to ping the IP from the Virtual machine Linux, but it doesn't work, even though I set an static IP in VMware network adapter VMnet1 . I also disabled the Firewall. And I used 0.0.0.0 as IP, but still the same problem appears. Maybe I set the IP in the wrong way?, or what could I have done wrong?

gavanderhoorn, 2019-02-21:
> I tried to ping the IP from the Virtual machine Linux, but it doesn't work, even though I set an static IP in VMware network adapter VMnet1 . 

So now you are using VMware?

Note: you cannot just change the IP of the VMware adapter. That is the adapter of the host *inside the vm*, not the IP of the Ubuntu guest.

> I also disabled the Firewall. And I used 0.0.0.0 as IP, but still the same problem appears. Maybe I set the IP in the wrong way?, or what could I have done wrong?

My advice would be to read up on VM configuration wrt networking.

Try using the *bridge* configuration as suggested by @gonzalocasas, and undo your changes to the VMware adapter.

nali12, 2019-02-21:
When I wanted to set an static IP in my Windows PC, I went to network connections. Here I found:
- Ethernet 
- Wifi (the one my Windows is connected)
- VMware Network Adapter 1
- VMware Network Adapter 8
I thought I could assign an IP inside VMware Network Adapter 1 , which i could use to refer to my robot in RobotStudio

gavanderhoorn, 2019-02-21:
So are you using VMware or Parallels? Those are two entirely different products.

nali12, 2019-02-21:
To make clear I have two computers. A mac running Parallels, inside installed Linux Ubuntu 14.06 and another computer Windows. Mac is connected to Ethernet and Windows PC to wifi.

gavanderhoorn, 2019-02-21:
And where is RobotStudio running? On the Windows machine?

Tbh this is really not an issue with `abb_driver`.

You'll have to make sure that your Windows machine *and* the Ubuntu guest are able to communicate with each other. Easiest would be to put them on the same network (ie: subnet). Routed would also be possible, but harder to configure.

A good test is being able to `ping` both hosts from each other. If that doesn't work, there's a high chance nothing else will work either.

> To make clear I have two computers. A mac running Parallels, inside installed Linux Ubuntu 14.06 and another computer Windows. Mac is connected to Ethernet and Windows PC to wifi.

I'm thoroughly confused by how changing the settings on the VMware adapter would ever work here, seeing as VMware is not used on either the Windows machine nor the Mac.

nali12, 2019-02-21:
Yes, RobotStudio is running in Windows. Indeed, I made a mistake changing the setting of the VMware adapter, that doesn't have anything to do with this.