ros-industrial/abb #177: can't locate node [robot_state] in package [abb_driver]

kahmed11, 2019-11-07:
Hello , am new with ROS and was trying to contact with the ABB irb_120 and with Robotstudio. In order to control the robot in RobotStudio I installed the ABB ROS server following all the tutorials ( http://wiki.ros.org/abb/Tutorials ). and when i run the command :
roslaunch abb_irb120_support robot_interface_download_irb120_3_58.launch robot_ip:=<ip_address>  i get always the error : 
ERROR: cannot launch node of type [abb_driver/robot_state]: can't locate node [robot_state] in package [abb_driver]
ERROR: cannot launch node of type [abb_driver/motion_download_interface]: can't locate node [motion_download_interface] in package [abb_driver]
if someone faced the same error or know why its there will be so thankful , wish u all a nice day

gavanderhoorn, 2019-11-07:
How did you install the `abb_driver` packages? Did you use `apt` or did you create a workspace and then `catkin_make`d them?

If the latter: please show us the exact commands you use to create, populate, build and activate your workspace.

gavanderhoorn, 2019-11-07:
And as I expect that there isn't actually anything wrong with the packages in this repository, I'm going to close this issue.

That does not mean you cannot continue discussing your problem here, of course. So please do.

closed 2019-11-07 by gavanderhoorn

kahmed11, 2019-11-07:
its working now  , i created a new workspace and its all good  , i think was something wrong with building the old workspace  . thanks  :)