ros-industrial/abb #178: REST based interface for ABB

johnwason, 2020-02-29:
Greetings

A while ago I developed a Python interface for an ABB that used the built in REST interface instead of running a socket server on the robot itself. It worked by modifying persistent variables, and was able to enter and exit EGM mode. It has a ROS interface for trajectory and other operations. I haven't tested it on the newer versions of the robot, but I'll point it out since it may be useful.

https://github.com/rpiRobotics/rpi_abb_irc5

gavanderhoorn, 2021-06-08:
Thanks for making this available @johnwason.

In the meantime ABB has released [abb_robot_driver](https://github.com/ros-industrial/abb_robot_driver) which comes with a similar node. It also uses RWS and EGM.

gavanderhoorn, 2021-06-08:
As this is not an issue with the packages in this repository, I'm going to close it.

closed 2021-06-08 by gavanderhoorn