ros-industrial/abb #185: Too many close points while executing in Robot Studio.

deep0105, 2021-01-18:
Hi,

I am using irb2600Ik plugin and descartes for planning motion through bunch of points. The planned motion seems to be executing perfectly on rviz but on Robot Studio it shows **Too many close positions.**

Thanks in advance.

gavanderhoorn, 2021-01-18:
This error (probably `50078 - too many closely spaced positions`) is shown on the TP when your trajectory points are too close together.

There isn't much we can do, apart from making your trajectory sparser.

This is a RAPID / ABB limitation, not of the driver, nor of descartes.

gavanderhoorn, 2021-01-18:
If you'd like more control over the motion, I'd recommend trying out [abb_robot_driver](https://github.com/ros-industrial/abb_robot_driver), which offers a much better external motion interface than the package in this repository.

closed 2021-01-18 by gavanderhoorn

gavanderhoorn, 2021-01-18:
Oh, and:

> The planned motion seems to be executing perfectly on rviz

please realise: RViz is not executing anything.

It's only visualising data. There is no simulation there.

deep0105, 2021-01-19:
Thanks @gavanderhoorn