ros-industrial/abb #186: Support package ABB GoFa™ CRB 15000

DavidJenny, 2021-06-08:
Is there already a plan in the pipeline to release a support package for CRB15000? A urdf would be quite helpful to start with.

gavanderhoorn, 2021-06-08:
What about https://github.com/ros-industrial/abb_experimental/pull/127?

DavidJenny, 2021-06-08:
Thanks a lot, wasn't aware of this!

closed 2021-06-08 by DavidJenny

gavanderhoorn, 2021-06-08:
If you could test it and verify it's correct, that would help getting it merged.

DavidJenny, 2021-06-08:
My plan is to work in simulation for now to see if it fits our application but I have a chance to test with the hardware this summer. Will keep you updated!

gavanderhoorn, 2021-06-08:
Checking it against RobotStudio and/or the specifications would already be valuable.

gavanderhoorn, 2021-07-13:
@DavidJenny: have you had a chance to test the package?