ros-industrial/abb #190: Dependency on abb_driver

gonzalocasas, 2022-04-01:
I know this is superfluous whining, but the dependency on `abb_driver` makes very little sense to me, if nobody is maintaining that package and there's no real dependency on it for the actual content of this repository that only has support packages but nothing related to control. Having this declared dependency means that after adding this repo to my catkin workspace I need to add `--skip-keys="abb_driver"` every time I run `rosdep`.

Feel free to close if you don't think it's worth the effort, but, also consider that the effort of getting rid of the dependency might be more worthy than the effort of holding it for backwards-compat reasons that don't really make much sense (the old driver is frankly un-usably limited).

gavanderhoorn, 2022-04-03:
This also came up over at `abb_experimental`: https://github.com/ros-industrial/abb_experimental/issues/129.

And nit-picking, but:

> there's no real dependency on it for the actual content of this repository that only has support packages but nothing related to control

is not entirely true.

All support packages come with `.launch` files which depend on `abb_driver`.

We can remove it all, but that will require *someone* to write documentation/tutorials to explain how users are supposed to create `.launch` files for their robots.

We see the same thing in `kuka_experimental`: there support packages do not come with any `.launch` files other than `load_...launch` and `test_...launch`, and every so often questions are posted / emailed about how to "control real robots".

gavanderhoorn, 2022-04-03:
And another related discussion: https://github.com/ros-industrial/ros_industrial_issues/issues/50 -- with a potential approach to at least notifying users as to *why* the provided `.launch` files don't work if they don't install `abb_driver` (or RViz in that case).

gonzalocasas, 2022-04-04:
I see. The options discussed in the rviz one are interesting. I would have assumed that adding a little comment to the launch file saying "` would suffice. I think the same applies here, this doesn't need to be overengineered, a simple note inside these files should be enough. But if you think that's not a viable option, feel free to close this.

gavanderhoorn, 2022-07-09:
I'll release `abb_driver` for Noetic.

That should take care of this.

I agree with you the dependency is unfortunate.

At least with any ROS 2 packages, we (I) won't be making that mistake again.

gavanderhoorn, 2022-09-01:
Done.

Should be part of the next sync.

closed 2022-09-01 by gavanderhoorn