ros-industrial/abb #202: Add support for 10kg and 12kg variants of GoFa robots (abb_crb15000_support)

gonzalocasas, 2024-07-10:
There are two more models on the GoFa product line for 10kg and 12kg payloads. It would be great to add support for them to this package. Is there a source from which we could get a hold of the URDF and meshes of those models? If so, I can prepare the PR to add it to this repo.

gonzalocasas, 2024-07-10:
/cc @mmmarcopalma

gavanderhoorn, 2024-07-10:
> Is there a source from which we could get a hold of the URDF and meshes of those models? If so, I can prepare the PR to add it to this repo.

The model I added in ros-industrial/abb_experimental#127 was based on a conversion from the CAD made available by ABB on their website.

If the variants you mention do not have any geometrical differences, you'd only have to add a new `xacro:macro`, reusing the original meshes, but (I assume) updated joint limits.

mmmarcopalma, 2024-07-10:
Thanks! I started updating the URDF you provided also converting from the CAD models.

> If the variants you mention do not have any geometrical differences, you'd only have to add a new `xacro:macro`, reusing the original meshes, but (I assume) updated joint limits.

The 3 GoFa models have geometrical differences (mainly links 2,3,4 have different lengths), so the meshes should be updated.   
How did you find joint limit values? Also from ABB data sheet?

gavanderhoorn, 2024-07-10:
> The 3 GoFa models have geometrical differences (mainly links 2,3,4 have different lengths), so the meshes should be updated.

Please make sure to reuse meshes as much as possible. If only those links changed, reuse the existing meshes for the other links.

> How did you find joint limit values? Also from ABB data sheet?

yes, that'd be a reliable source of information, although in case of asymmetric limits, you'll either need to check them against a real robot, or RobotStudio.

(technically you could also use RWS to get the info, as `abb_robot_driver` does, but that's probably more work)

gavanderhoorn, 2025-04-05:
Just drawing attention to the fact this repository is only *community supported*, so support for the other/missing CRB 15000 variants will have to be contributed.